google/android-riscv64 #117: VTS can't work.

nylon7, 2023-11-02:
I use VTS in my environment and currently encounter some problems.

- Unable to find `aapt` in path.
```
$ vts-traded
Unable to find `aapt` in path
```
We need to change the directory separately to compile it under aapt to solve this problem
```
$cd frameworks/base/tools/aapt
$mma
```
- Cannot directly use vts-tradefed for verification
```
nylonc@hsinchu26:/scratch2/nylonc/android_up_aosp$ vts-tradefed
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV6_0GeneratorTest/riscv64/tests/apm_config_b_205808571_6_0.xml d$
esn't look like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV6_0GeneratorTest/riscv64/tests/apm_config_no_vx.xml doesn't loo$
 like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV6_0GeneratorTest/riscv64/tests/apm_config_with_vx.xml doesn't l$
ok like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV7_0GeneratorTest/riscv64/tests/apm_config_b_204314749_7_0.xml do
esn't look like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV7_0GeneratorTest/riscv64/tests/apm_config_b_205808571_7_0.xml do
esn't look like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV7_0GeneratorTest/riscv64/tests/apm_config_no_vx_7_0.xml doesn't
look like a test configuration.
05:15:21 E/ConfigurationUtil: /scratch2/nylonc/android_up_aosp/out/target/product/vsoc_riscv64/testcases/HalAudioV7_0GeneratorTest/riscv64/tests/apm_config_with_vx_7_0.xml doesn'
t look like a test configuration.
==================
Notice:
We collect anonymous usage statistics in accordance with our Content Licenses (https://source.android.com/setup/start/licenses), Contributor License Agreement (https://opensource.g
oogle.com/docs/cla/), Privacy Policy (https://policies.google.com/privacy) and Terms of Service (https://policies.google.com/terms).
==================
Android Vendor Test Suite 14_r1 (eng.nylonc.20231031.020405)
Use "help" or "help all" to get more information on running commands.
11-01 05:15:21 D/NativeDevice: Device 0.0.0.0:6520 state is now NOT_AVAILABLE
11-01 05:15:21 D/NativeDevice: Device 0.0.0.0:6520 state is now ONLINE
11-01 05:15:27 I/DeviceManager: Detected new device 0.0.0.0:6520
vts-tf > run vts-kernel -m vts_linux_kselftest_riscv_64
11-01 05:15:39 I/TestInvocation: Starting invocation for 'vts-kernel' with '[ DeviceBuildInfo{bid=eng.nylonc.20231031.020405, serial=0.0.0.0:6520} on device '0.0.0.0:6520']
11-01 05:15:40 W/NativeDevice: Failed to query battery level for 0.0.0.0:6520: java.util.concurrent.TimeoutException: Waited 500 milliseconds (plus 75977 nanoseconds delay) for Set
tableFuture@549b2550[status=PENDING]
11-01 05:15:40 W/NativeDevice: Failed to query battery level for 0.0.0.0:6520: java.util.concurrent.TimeoutException: Waited 500 milliseconds (plus 68253 nanoseconds delay) for Set
tableFuture@549b2550[status=PENDING]
11-01 05:15:42 W/PropertyFetcher: ShellCommandUnresponsiveException getting properties for device 0.0.0.0:6520
java.lang.Throwable: ShellCommandUnresponsiveException getting properties for device 0.0.0.0:6520
        at com.android.ddmlib.PropertyFetcher.handleException(PropertyFetcher.java:248)
        at com.android.ddmlib.PropertyFetcher$1.run(PropertyFetcher.java:211)
Caused by: com.android.ddmlib.ShellCommandUnresponsiveException
        at com.android.ddmlib.AdbHelper.executeRemoteCommand(AdbHelper.java:731)
        at com.android.ddmlib.AdbHelper.executeRemoteCommand(AdbHelper.java:511)
        at com.android.ddmlib.internal.DeviceImpl.executeShellCommand(DeviceImpl.java:722)
        at com.android.ddmlib.PropertyFetcher$1.run(PropertyFetcher.java:207)
11-01 05:15:44 W/BatteryFetcher: ShellCommandUnresponsiveException getting battery level for device 0.0.0.0:6520: null
vts-tf >
vts-tf > run vts-kernel -m vts_linux_kselftest
11-01 05:16:16 D/CommandScheduler: No available device matching all the config's requirements for cmd id 2.
Command will be rescheduled: [vts-kernel, -m, vts_linux_kselftest]
11-01 05:22:13 W/NativeDeviceStateMonitor: Device 0.0.0.0:6520 did not boot after 360000 ms
11-01 05:26:30 W/NativeDeviceStateMonitor: Device 0.0.0.0:6520 did not boot after 240000 ms
11-01 05:30:35 W/NativeDeviceStateMonitor: Device 0.0.0.0:6520 did not boot after 240000 ms
11-01 05:30:40 W/TestInvocation: Invocation did not complete due to device 0.0.0.0:6520 becoming not available. Reason: com.android.tradefed.device.DeviceUnresponsiveException[DEVI
CE_UNRESPONSIVE|520751|LOST_SYSTEM_UNDER_TEST]: Device 0.0.0.0:6520 is online but unresponsive
11-01 05:35:46 W/NativeDevice: Command: 'shell bugreportz' on '0.0.0.0:6520' went over its timeout for outputing a response.
11-01 05:35:46 E/TestDevice: Something went went wrong during bugreportz collection: ''
11-01 05:35:46 W/TestInvocation: Error when collecting bugreport for device '0.0.0.0:6520'
11-01 05:35:46 E/RunCommandTargetPreparer: Skipping command teardown since exception was DeviceNotAvailable
11-01 05:35:46 E/RunCommandTargetPreparer: Skipping command teardown since exception was DeviceNotAvailable
11-01 05:35:46 E/TestInvocation: Counting as virtual_device_lost.
11-01 05:35:46 W/NativeDevice: Attempting to stop logcat when not capturing for 0.0.0.0:6520
11-01 05:35:46 I/SuiteResultReporter:
============================================
================= Results ==================
=============== Consumed Time ==============
Total aggregated tests run time: 0 ms
=============== Summary ===============
Total Run time: 20m 6s
0/0 modules completed
Total Tests       : 0
PASSED            : 0
FAILED            : 0
============== End of Results ==============
============================================
...
...
```
Judging from the results report, it did not execute any test programs, but it did take about 20 minutes to display this report.

In order to check whether it is a problem with the test program itself. 

```
# build machine
cd $ROOT_PATH/external/linux-kselftest
mma
adb connect 192.168.51.14
adb sync data

# aosp shell
adb shell
cd /data/nativetest64/linux-kselftest
# example
cd exec
./execveat.sh
```
It works fine.

So I guess there is something wrong with the VTS framework itself.

enh-google, 2023-11-03:
> We need to change the directory separately to compile it under aapt to solve this problem

that seems unlikely? doesn't a top-level build get you an aapt binary? it does for me:
```
~/aosp-main-with-phones$ find out/host/ -name aapt
out/host/linux-x86/bin/aapt
```

> Cannot directly use vts-tradefed for verification

(i'm not aware that anyone at google has even tried to run VTS yet.)

nylon7, 2023-11-06:
Hi @enh-google 

> that seems unlikely? doesn't a top-level build get you an aapt binary?

I'm not sure, Maybe it's a problem with my environment.

> (i'm not aware that anyone at google has even tried to run VTS yet.)

Will your internal CI testing include VTS in the test project?

prashanthswami, 2023-11-10:
If I recall correctly, you need to build VTS in a particular way to make sure all the relevant tools are sourced. Are you building VTS like so: https://source.android.com/docs/core/tests/vts/systems ? While VTS in CI is probably going to be a while away, it does work locally on AOSP and we've tried the standard kselftests.

nylon7, 2023-11-14:
Hi @prashanthswami this is my steps
1. unch aosp_cf_riscv64_phone-trunk_staging-userdebug
2. make vts
3. launch_cvd -cpus=4 --memory_mb=16384
4. adb wait-for-device
5. vts-tradefed
6. run vts-kernel -m vts_linux_kselftest
Is there something missing in my steps?

nylon7, 2023-12-06:
Hi @prashanthswami , Is there any part I've overlooked?

prashanthswami, 2024-02-24:
Apologies, not self-assigning this dropped it off my radar - assuming you're still having this problem, I don't see anything wrong with your steps. In fact, running them myself, this works fine.

As a baseline, can you try syncing the tree and confirming it's still doing the same thing? If you're seeing TradeFed complain the device is unresponsive, my suspicion is that the device is just truly running too slowly. If you shell into the device and run a command like 'dumpsys', is that command hanging a bunch? I'm trying to get a sense of how slow it's running on your host.

nylon7, 2024-02-26:
Hi @prashanthswami , I will try again thanks for your feedback.

closed 2024-11-12 by enh-google