google/android-riscv64 #120: Implement System.arraycopy intrinsic

alpencolt, 2023-11-03:
Need to implement:

- [ ] `SystemArrayCopy`
- [ ] `SystemArrayCopyChar`
- [ ] `SystemArrayCopyByte`
- [ ] `SystemArrayCopyInt`

@enh-google could you assign issue to me

alpencolt, 2023-11-15:
What the proper way for testing intrinsic implementation before submitting commit? I use
```
./art/test/testrunner/testrunner.py --target --no-prebuild  --jit-on-first-use -j 8 --64
```
Is it enough?

enh-google, 2023-11-16:
i asked the ART folks, and one said:

> I think using AOT would be better than JIT but you could also run both with `testrunner.py --target --64 --ndebug --optimizing --jit`.
> 
> Just to clarify, the way to use `testrunner` on `device` is to use the [chroot method](https://cs.android.com/android/platform/superproject/main/+/main:art/test/README.chroot.md).

while another said:

> I usually test only `--optimizing` but if there is some JIT-specific code, I also test with `--jit`. I must admit that I'm not using `--jit-on-first-use` but it may actually be better than plain `--jit` in most cases (except when testing inline caches and other stuff that needs to warm up).
> 
> On riscv64, we currently have
>   - two failures with `--optimizing`: 137-cfi (needs investigation) and 640-checker-integer-valueof (missing Integer.valueOf() intrinsic).
>   - four failures with `--jit`: 137-cfi64, 845-data-image, 846-multidex-data-image, 913-heaps (all need investigation)

alpencolt, 2023-11-17:
@enh-google thank you!
Yes, I use chroot based testing on cuttlefisg, but have more failures even without my changes, probably because of have not latest source code.
I'm going to rebase and check it with different plans.

enh-google, 2023-11-17:
(yeah, it's always worth running the tests _before_ for this exact reason, especially with riscv64 where the status basically changes from day to day!)

alpencolt, 2024-02-01:
`SystemArrayCopyChar/Byte/Int` implementation: https://android-review.googlesource.com/c/platform/art/+/2939599

closed 2025-05-29 by enh-google