google/android-riscv64 #122: Implement RISC-V Vector extension instructions in MacroAssembler

romart, 2023-11-15:
There is released [spec](https://github.com/riscv/riscv-v-spec/releases/tag/v1.0) about RVV which needs to be implemented

romart, 2023-11-16:
@enh-google Could you please assign this issue to me

romart, 2023-11-30:
@enh-google Hi, I see there is [test file](https://android.googlesource.com/platform/art/+/refs/heads/main/compiler/utils/riscv64/assembler_riscv64_test.cc) for riscv assembler but it is not clear how to run those tests. Could you please clarify a few questions for me? 

1. Where are those tests supposed to be run? AFAIU they require clang
2. How to build those tests?
3. How to run them on host?

enh-google, 2023-12-06:
one of the ART developers said:

For gtests in general:
* For host follow [this](https://cs.android.com/android/platform/superproject/main/+/main:art/test/README.md)
* For device follow [this](https://cs.android.com/android/platform/superproject/main/+/main:art/test/README.chroot.md).

That test in particular is part of `art_compiler_tests`. In the readme there are ways to filter and run a subset of tests.

-------------

one of the other ART developers pointed out that there's also
```
    m test-art-host-gtest
```

As for the development and testing tree, I'm using `master-art`. After a long time of explicitly following the
    art/test/[README.chroot.md](http://readme.chroot.md/)
setup and manually executing 6 commands, I've created a helper function in my `.bashrc`:
    function my_unbundled_lunch()
    {
        export SOONG_ALLOW_MISSING_DEPENDENCIES=true
        export TARGET_BUILD_UNBUNDLED=true
        export BUILD_BROKEN_DISABLE_BAZEL=true
        . build/envsetup.sh
        lunch $@
        export PATH="$(pwd)/prebuilts/runtime:$PATH"
        export ADB="$ANDROID_BUILD_TOP/prebuilts/runtime/adb"
    }
and the full setup and test would be
    # Create and sync the tree in a new directory
    repo init -u sso://[android.googlesource.com/platform/manifest](http://android.googlesource.com/platform/manifest) -b master-art  # Non-googlers may need a different protocol, maybe `persistent-https://`.
    repo sync -c -j 4  # This can take a long time. You may want to re-sync afterwards, maybe with `-j 40`.
    # Setup
    my_unbundled_lunch riscv64-trunk_staging-eng  # Or explicitly executing the 6 commands from that function.
    // Build and test
    m test-art-host-gtest

For quick iteration on a particular host test (or set of host tests):
    m out/host/linux-x86/nativetest64/art_compiler_host_tests/art_compiler_host_tests && out/host/linux-x86/nativetest64/art_compiler_host_tests/art_compiler_host_tests --gtest_filter='AssemblerRISCV64Test.*'
where I change the `--gtest_filter` based on my current needs. (Note that `AssemblerRISCV64Test.*` is actually a lot of tests, so I'd recommend a smaller set.)
If I need to debug a particular test with `gdb`, I use
    gdb --args out/host/linux-x86/nativetest64/art_compiler_host_tests/art_compiler_host_tests --gtest_filter='AssemblerRISCV64Test.JMaxOffset21BackwardBare' --no_isolate
Sometimes (rarely) the 32-bit test behaves differently, so you may need to use `nativetest` instead of `nativetest64`.

romart, 2023-12-07:
@enh-google  Thank you for you response

I finally was able to implement, build and run tests by myself

The Gerrit review is [here](https://android-review.googlesource.com/c/platform/art/+/2862810/), contains of 4 commits

enh-google, 2024-01-12:
i talked to the ART folks again about testing, and they confirmed that they're just running the ART tests atm because everything else is so slow. in fact, they pointed out you might have to run the ART tests multiple times to avoid timeouts, or increase some of the timeouts (details below)...

"""
I'd recommend running ART tests on the device as described in the `README.chroot.md`. At the very least
    art/tools/run-gtests.sh
    art/test/testrunner/testrunner.py --target --64 --optimizing --jit --interpreter

When testing on an Ubuntu VM (without V), run-tests require `--ndebug` as they do not even start properly with the default `--debug`.

I currently see one gtest failure (ArtExecTest.DropCapabilities) and the `137-cfi` failure for `--optimizing`. The `--jit` and `--interpreter` tests have a few more failures in addition to the `137-cfi`.

When testing on the riscv64 cuttlefish, I additionally see a lot of timeouts (gtest and run-tests, both `--debug` and `--ndebug`) and some non-timeout gest failures (to be investigated).

Some of these timeouts disappear if I run tests individually but running the entire suite with `-j 1` would be quite time consuming but I've started that and hope that the cuttlefish is stable enough to survive over the weekend.
"""

alpencolt, 2024-01-14:
@enh-google thank you for sharing!

romart, 2024-06-11:
RISC-V Vector extension support is merged into upstream

https://android-review.googlesource.com/c/platform/art/+/2862810

Task is done

closed 2024-06-11 by romart