google/android-riscv64 #123: Seems like cuttlefish does not work

romart, 2023-11-22:
Trying to run emulator using

```
$ launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=none \
      --vm_manager=qemu_cli \
      --qemu_binary_dir=$ANDROID_BUILD_TOP/device/google/cuttlefish_vmm/qemu/x86_64-linux-gnu/bin
```

But if fails with the following 

```
roman@rya-sc:~/git/android$ launch_cvd -cpus=4 --memory_mb=8192 --qemu_binary_dir=/home/roman/qemu/android/riscv/build
exe E 11-22 18:18:27  3090  3099 start.cpp:475] launcher command: HOME="/tmp/1000/cvd_1" ANDROID_HOST_OUT="/home/roman/git/android/out/host/linux-x86" ANDROID_SOONG_HOST_OUT="/home/roman/git/android/out/host/linux-x86" ANDROID_PRODUCT_OUT="/home/roman/git/android/out/target/product/vsoc_riscv64" CUTTLEFISH_INSTANCE="1"  /home/roman/git/android/out/host/linux-x86/bin/cvd_internal_start -cpus=4 --memory_mb=8192 --qemu_binary_dir=/home/roman/qemu/android/riscv/build --webrtc_device_id=cvd_1-1 --base_instance_num=1 --group_id=cvd_1
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services.
===================================================================

11-22 18:18:27.581  5949  5949 D assemble_cvd: assemble_cvd.cc:91 Could not locate fetcher config file.
11-22 18:18:27.582  5949  5949 I assemble_cvd: config_flag.cpp:146 Launching CVD using --config='phone'.
11-22 18:18:27.584  5949  5949 D assemble_cvd: subprocess.cpp:435 Started (pid: 5952): /home/roman/git/android/out/host/linux-x86/bin/extract-ikconfig
11-22 18:18:27.584  5949  5949 D assemble_cvd: subprocess.cpp:437 /home/roman/git/android/out/target/product/vsoc_riscv64/boot.img
tail: error writing 'standard output': Broken pipe
11-22 18:18:27.662  5949  5949 D assemble_cvd: subprocess.cpp:435 Started (pid: 5960): /home/roman/git/android/out/host/linux-x86/bin/avbtool
11-22 18:18:27.662  5949  5949 D assemble_cvd: subprocess.cpp:437 info_image
11-22 18:18:27.662  5949  5949 D assemble_cvd: subprocess.cpp:437 --image
11-22 18:18:27.662  5949  5949 D assemble_cvd: subprocess.cpp:437 /home/roman/git/android/out/target/product/vsoc_riscv64/boot.img
Enabling --gpu_mode=drm_virgl.
Requested resuming a previous session (the default behavior) but the base images have changed under the overlay, making the overlay incompatible. Wiping the overlay files.
Path for instance UDS: /tmp/cf_avd_1000
The following lines contain useful debugging information:
  Serial console is disabled; use -console=true to enable it.
  Logcat output: /tmp/1000/cvd_1/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /tmp/1000/cvd_1/cuttlefish/instances/cvd-1/kernel.log
  Launcher log: /tmp/1000/cvd_1/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /tmp/1000/cvd_1/cuttlefish/instances/cvd-1/cuttlefish_config.json
  Launcher Build ID: eng.roman.20231115.164729
Failed to read a complete exit code, read 0 bytes only instead of the expected 4
VIRTUAL_DEVICE_BOOT_FAILED
11-22 18:18:37.991  5925  5925 E cvd_internal_start: main.cc:465 run_cvd returned 10
3. main.cc:130 | CvdMain |
2. client.h:67 | HandleCommand |
1. client.cpp:326 | HandleCommand |
 | device/google/cuttlefish/host/commands/cvd/client.cpp:409
 v Result<void> cuttlefish::CvdClient::CheckStatus(const cvd::Status &, const std::string &)
Received error response for "HandleCommand":
Exited with code 255
In client
```

My system is 
```
roman@rya-sc:~/git/android$ uname -a
Linux rya-sc 6.1.0-13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux
```

Is that known issue or do I make something wrong?

gongwuchun, 2023-11-23:
you could check logcat log(cuttlefish/instances/cvd-1/logs/logcat) if zygote failed like following, and it should be fixed by https://android-review.googlesource.com/c/platform/art/+/2845177

01-01 00:06:06.718  1289  1289 E Zygote  : System zygote died with fatal exception
01-01 00:06:06.718  1289  1289 E Zygote  : java.lang.ExceptionInInitializerError
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundleReader$ReaderCache.createInstance(ICUResourceBundleReader.java:192)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundleReader$ReaderCache.createInstance(ICUResourceBundleReader.java:181)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.SoftCache.getInstance(SoftCache.java:71)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundleReader.getReader(ICUResourceBundleReader.java:236)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle.createBundle(ICUResourceBundle.java:1516)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle$5.load(ICUResourceBundle.java:1387)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle$1.createInstance(ICUResourceBundle.java:98)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle$1.createInstance(ICUResourceBundle.java:95)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.SoftCache.getInstance(SoftCache.java:71)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle.instantiateBundle(ICUResourceBundle.java:1374)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle.getBundleInstance(ICUResourceBundle.java:1259)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUResourceBundle.getBundleInstance(ICUResourceBundle.java:1235)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.util.UResourceBundle.getRootType(UResourceBundle.java:321)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.util.UResourceBundle.instantiateBundle(UResourceBundle.java:357)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.util.UResourceBundle.getBundleInstance(UResourceBundle.java:147)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.util.UResourceBundle.getBundleInstance(UResourceBundle.java:110)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.NumberingSystem.lookupInstanceByName(NumberingSystem.java:229)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.NumberingSystem.<clinit>(NumberingSystem.java:43)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols.loadData(DecimalFormatSymbols.java:1394)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols.-$$Nest$smloadData(Unknown Source:0)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols$1.createInstance(DecimalFormatSymbols.java:1893)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols$1.createInstance(DecimalFormatSymbols.java:1890)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.SoftCache.getInstance(SoftCache.java:71)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols.getCachedLocaleData(DecimalFormatSymbols.java:1346)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols.initialize(DecimalFormatSymbols.java:1358)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.text.DecimalFormatSymbols.<init>(DecimalFormatSymbols.java:71)
01-01 00:06:06.718  1289  1289 E Zygote  :      at com.android.i18n.system.ZygoteHooks.onBeginPreload(ZygoteHooks.java:60)
01-01 00:06:06.718  1289  1289 E Zygote  :      at dalvik.system.ZygoteHooks.onBeginPreload(ZygoteHooks.java:70)
01-01 00:06:06.718  1289  1289 E Zygote  :      at com.android.internal.os.ZygoteInit.beginPreload(ZygoteInit.java:167)
01-01 00:06:06.718  1289  1289 E Zygote  :      at com.android.internal.os.ZygoteInit.preload(ZygoteInit.java:128)
01-01 00:06:06.718  1289  1289 E Zygote  :      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:831)
01-01 00:06:06.718  1289  1289 E Zygote  : Caused by: java.lang.IndexOutOfBoundsException: index=2 out of bounds (limit=0)
01-01 00:06:06.718  1289  1289 E Zygote  :      at java.nio.Buffer.checkIndex(Buffer.java:756)
01-01 00:06:06.718  1289  1289 E Zygote  :      at java.nio.DirectByteBuffer.get(DirectByteBuffer.java:253)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUBinary.readHeader(ICUBinary.java:590)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUBinary$DatPackageReader.validate(ICUBinary.java:61)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUBinary.addDataFilesFromFolder(ICUBinary.java:359)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUBinary.addDataFilesFromPath(ICUBinary.java:325)
01-01 00:06:06.718  1289  1289 E Zygote  :      at android.icu.impl.ICUBinary.<clinit>(ICUBinary.java:301)
01-01 00:06:06.718  1289  1289 E Zygote  :      ... 31 more
01-01 00:06:06.719  1289  1289 D AndroidRuntime: Shutting down VM

romart, 2023-11-23:
@gongwuchun Unfortunately there is no that log in `cuttlefish/instances/cvd-1/logs/` so probably this is a different issue

gongwuchun, 2023-11-23:
@romart any other logs under cuttlefish/instances/cvd-1/logs/

you can have a try build qemu by yourself, btw, update your AOSP to latest.
`sudo apt-get install -y ninja-build libpixman-1-dev build-essential libsdl2-image-dev libgtk-3-dev libslirp-dev libnuma-dev libusb-1.0-0-dev libusbredirparser-dev libvirglrenderer-dev libepoxy-dev libgbm-dev
git clone https://github.com/qemu/qemu.git
cd qemu
./configure --target-list=riscv64-softmmu,riscv64-linux-user --enable-opengl --enable-virglrenderer --enable-sdl --enable-sdl-image --enable-gtk --enable-slirp --enable-libusb --enable-numa --enable-usb-redir
make -j $(nproc)`

romart, 2023-11-23:
@gongwuchun there is `launcher.log` only and it contains `stdout`, nothing useful 

What's wrong with my qemu?

```
roman@rya-sc:~/git/android$ qemu-system-riscv64 --version
QEMU emulator version 8.1.50
Copyright (c) 2003-2023 Fabrice Bellard and the QEMU Project developers
```

romart, 2023-11-23:
Actually brand-new qemu built from git works for some reason

enh-google, 2023-11-27:
> Actually brand-new qemu built from git works for some reason

that might be because (as @gongwuchun pointed out and fixed with https://android-review.googlesource.com/c/platform/bionic/+/2844983) i got the test for "if we're on qemu, disable V" the wrong way round, and you're having fewer/no V issues with ToT qemu?

anyway, it sounds like this is working again now, and can be closed? (file a new bug if there are remaining issues that need to be looked at.)

closed 2023-11-27 by enh-google

irfanwaheed-WF, 2023-11-27:
is it recommended to use ToT QEMU or the prebuilt QEMU?

enh-google, 2023-11-28:
> is it recommended to use ToT QEMU or the prebuilt QEMU?

"all the options are bad". right now, i've actually been doing fine with the current debian testing qemu, but @prashanthswami has been having all kinds of crashes (including with ToT, https://gitlab.com/qemu-project/qemu/-/issues/1976). that's why there's the change to disable bionic's V string/memory routines for qemu atm (and also why i needed @gongwuchun to point out that i'd gotten it backwards!). we're still working on getting a good qemu that "just works", at which point we'll make launch_cvd default to that. but for now it's more a case of "which qemu bugs are going to annoy you the least?" :-(