google/android-riscv64 #126: ART: Implement "C" extension instructions in the `Riscv64Assembler`

enh-google, 2023-12-06:
For code size reduction, it would be nice for the existing instruction functions to just emit the "C" versions if available for the given arguments. However, for certain things (thinking of the introspection Baker read barrier on arm/arm64), it would be nice to have the ability to precisely specify whether to use a 16-bit or 32-bit instruction. So, I would like to see explicit 16-bit instruction functions (say, `CLdsp()`) and a flag whether to use them automatically (say, in `Ld()`) and a way to disable the flag (say, `class ScopedNoCInstructions`).

This should include adding assembler tests and changing the `AssemblerRISCV64Test` so that we can run tests with and without the "C" extension. (Maybe define a parametrized test class and change appropriate tests from `TEST_F`s to `TEST_P` with an argument whether to use the "C" extension or not.)

Note that this can be split into several changes as we do not need to add all "C" instructions together.

romart, 2023-12-20:
@enh-google could you please assign this issue to me?

romart, 2024-02-01:
Gerrit review 
https://android-review.googlesource.com/c/platform/art/+/2939939
https://android-review.googlesource.com/c/platform/art/+/2939940
https://android-review.googlesource.com/c/platform/art/+/2939941
https://android-review.googlesource.com/c/platform/art/+/2939942

romart, 2024-06-11:
Finally merged into  upstream 
https://android-review.googlesource.com/c/platform/art/+/3000792

closed 2024-06-11 by enh-google