google/android-riscv64 #128: ART: implement String intrinsics

enh-google, 2023-12-06:
There are some `String` intrinsics to implement, most importantly `StringCompareTo` and `StringEquals`.

The question is whether we want to bother with scalar implementations or implement them with the "V" extension once we have the assembler support.

(bionic only has V-optimized implementations of string/memory functions.)

GreenSeal, 2023-12-22:
Hello, @enh-google !
Could you please assign this issue to me?

prashanthswami, 2024-02-24:
Work is ongoing in [this change](https://android-review.googlesource.com/c/platform/art/+/2958172).

GreenSeal, 2024-04-02:
Both intrinsics are successfuly merged
StringEquals: https://android-review.googlesource.com/c/platform/art/+/2958172
StringCompareTo: https://android-review.googlesource.com/c/platform/art/+/3001232

enh-google, 2024-04-02:
there are still three other unimplemented String intrinsics in the list?
```
  V(StringGetCharsNoCheck)                      \
  V(StringStringIndexOf)                        \
  V(StringStringIndexOfAfter)                   \
```
although it looks like arm64/x86-64 don't implement the latter two either?

we'll probably want to come back to StringEquals and StringCompareTo and switch them over to V when qemu is fixed too?

GreenSeal, 2024-04-02:
I'm working on StringStringIndexOf and Mark Gonopolskiy had already opened MR on StringGetChatsNoCheck
https://android-review.googlesource.com/c/platform/art/+/3005208
Yes, there is no arm64/x86-64 versions of the latter two instrinsics
I think, switching them over "V" extension could make significant impact on these instrinsics

enh-google, 2025-05-29:
(i'll file a new bug for "revisit ART intrinsics to use V".)

closed 2025-05-29 by enh-google