google/android-riscv64 #129: lldb server not found

wzhbest, 2023-12-20:
Based on the experiment of LLDB debugging in Android Studio, it was found that the LLDB server cannot be found for the RISCV architecture. Is there a need for additional support
[image]

pirama-arumuga-nainar, 2023-12-20:
Duplicate fo #21 

NDK r27 will be the first release to have lldb-server for riscv64.  If you want to test, this is currently available in the NDK canary bulds: https://ci.android.com/builds/branches/aosp-master-ndk/grid?legacy=1

pirama-arumuga-nainar, 2023-12-20:
Duplicate of #21

closed 2023-12-20 by pirama-arumuga-nainar

MaoHan001, 2023-12-21:
> Duplicate of #21
> 
> NDK r27 will be the first release to have lldb-server for riscv64. If you want to test, this is currently available in the NDK canary bulds: https://ci.android.com/builds/branches/aosp-master-ndk/grid?legacy=1

I suppose this might be two separate issues, one with the NDK lacking the lldb-server and the other with the log showing that riscv64 cannot be recognized by getClientABIs/DebugProcessStarter.kt.
2023-12-19 15:37:57,330 [2058581] WARN - getClientABIs - Failed to get abi by name: riscv64
2023-12-19 15:37:57,330 [2058581] INFO - getClientABIs - ABIs supported by app: []
2023-12-19 15:37:57,334 [2058585] INFO - #c.i.e.r.ExecutionUtil - Error running 'app':
LLDB server for architectures [] not found
com.intellij.execution.ExecutionException: LLDB server for architectures [] not found
at com.android.tools.ndk.run.lldb.DebugProcessStarterKt.getDebugProcessStarter(DebugProcessStarter.kt:160)
at com.android.tools.ndk.run.lldb.DebugProcessStarterKt.getDebugProcessStarter(DebugProcessStarter.kt:79)
at com.android.tools.ndk.run.editor.NativeAndroidDebugger.getDebugProcessStarterForNewProcess(NativeAndroidDebugger.java:156)
at com.android.tools.ndk.run.editor.NativeAndroidDebugger.getDebugProcessStarterForNewProcess(NativeAndroidDebugger.java:60)
Even using canary builds with LLDB included doesn't seem to work properly.

wzhbest, 2023-12-22:
Using this [version ](ttps://ci.android.com/builds/branches/aosp-master-ndk/grid?legacy=1)[ttps://ci.android.com/builds/branches/aosp-master-ndk/grid?legacy=1](https://ci.android.com/builds/branches/aosp-master-ndk/grid?legacy=1) for verification, neither lldb nor profile works.

[idea.log](https://github.com/google/android-riscv64/files/13749421/idea.log)

lldb server not found:
[image]
[image]
 
Please help continue to take a look

enh-google, 2024-01-03:
what does `adb shell getprop | grep ro.product.cpu` show?

wzhbest, 2024-01-10:
please see the result:
[image]