google/android-riscv64 #131: Can' boot about

luyahan, 2024-01-02:
luyahan@luyahan-MIX:~/cf$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

**transport message failed, response body: device '0.0.0.0:6520' not found**

log:
```
luyahan@luyahan-MIX:~/cf$ HOME=$PWD ./bin/launch_cvd  --vm_manager=qemu_cli --gpu_mode=none
exe I 01-02 19:42:06  5967  5975 start.cpp:491] launcher command: HOME="/home/luyahan/cf" ANDROID_HOST_OUT="/home/luyahan/cf" ANDROID_SOONG_HOST_OUT="/home/luyahan/cf" ANDROID_PRODUCT_OUT="/home/luyahan/cf" CUTTLEFISH_INSTANCE="1"  /home/luyahan/cf/bin/cvd_internal_start --vm_manager=qemu_cli --gpu_mode=none --webrtc_device_id=cvd_1-1 --base_instance_num=1 --group_id=cvd_1
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services. You can adjust the permission for sending
diagnostic information to Google, such as crash reports and usage
data from this Android Virtual Device, at any time by running
"launch_cvd -report_anonymous_usage_stats=n"
===================================================================

01-02 19:42:06.675  6952  6952 D assemble_cvd: assemble_cvd.cc:91 Could not locate fetcher config file.
01-02 19:42:06.675  6952  6952 I assemble_cvd: config_flag.cpp:184 Reading --config option from: /home/luyahan/cf/android-info.txt
01-02 19:42:06.675  6952  6952 I assemble_cvd: config_flag.cpp:148 Launching CVD using --config='phone'.
01-02 19:42:06.676  6952  6952 D assemble_cvd: subprocess.cpp:435 Started (pid: 6955): /home/luyahan/cf/bin/extract-ikconfig
01-02 19:42:06.676  6952  6952 D assemble_cvd: subprocess.cpp:437 /home/luyahan/cf/boot.img
tail: error writing 'standard output': Broken pipe
01-02 19:42:06.742  6952  6952 D assemble_cvd: subprocess.cpp:435 Started (pid: 6963): /home/luyahan/cf/bin/avbtool
01-02 19:42:06.742  6952  6952 D assemble_cvd: subprocess.cpp:437 info_image
01-02 19:42:06.742  6952  6952 D assemble_cvd: subprocess.cpp:437 --image
01-02 19:42:06.742  6952  6952 D assemble_cvd: subprocess.cpp:437 /home/luyahan/cf/boot.img
GPU vhost user auto mode: not yet supported with qemu_cli. Not enabling vhost user gpu.
Path for instance UDS: /tmp/cf_avd_1000
The following lines contain useful debugging information:
  Serial console is disabled; use -console=true to enable it.
  Logcat output: /home/luyahan/cf/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /home/luyahan/cf/cuttlefish/instances/cvd-1/kernel.log
  Launcher log: /home/luyahan/cf/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/luyahan/cf/cuttlefish/instances/cvd-1/cuttlefish_config.json
  Launcher Build ID: P67223518
/home/luyahan/cf/bin/config_server
/home/luyahan/cf/bin/modem_simulator
/home/luyahan/cf/bin/tombstone_receiver
/home/luyahan/cf/bin/casimir_control_server
/home/luyahan/cf/bin/screen_recording_server
/home/luyahan/cf/bin/adb_connector
/home/luyahan/cf/bin/socket_vsock_proxy
/home/luyahan/cf/bin/socket_vsock_proxy
/home/luyahan/cf/bin/tcp_connector
/home/luyahan/cf/bin/tcp_connector
/home/luyahan/cf/bin/control_env_proxy_server
/home/luyahan/cf/bin/echo_server
/home/luyahan/cf/bin/gnss_grpc_proxy
/home/luyahan/cf/bin/logcat_receiver
/home/luyahan/cf/bin/kernel_log_monitor
/home/luyahan/cf/bin/metrics
/home/luyahan/cf/bin/openwrt_control_server
/home/luyahan/cf/bin/log_tee
/home/luyahan/cf/bin/pica
/home/luyahan/cf/bin/log_tee
/home/luyahan/cf/bin/process_restarter
Failed to open socket: Connection refused
/home/luyahan/cf/bin/netsimd
/home/luyahan/cf/bin/socket_vsock_proxy
/home/luyahan/cf/bin/socket_vsock_proxy
Failed to open socket: Connection refused
/home/luyahan/cf/bin/secure_env
/home/luyahan/cf/bin/qemu-system-riscv64
Start modem simulator, server_fds: 72, Sim type: normal
Metrics started
Metrics host received: VMStop
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/EchoServer.sock
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/CasimirControlServer.sock
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/ScreenRecordingServer.sock
Server listening on 0.0.0.0:7200
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/OpenwrtControlServer.sock
vector version is not specified, use the default value v1.0
vector version is not specified, use the default value v1.0
root-canal I 01-02 19:42:07.395 device_boutique.cc:39                    Registering beacon
root-canal I 01-02 19:42:07.395 device_boutique.cc:39                    Registering beacon_swarm
root-canal I 01-02 19:42:07.395 device_boutique.cc:39                    Registering scripted_beacon
root-canal I 01-02 19:42:07.395 device_boutique.cc:39                    Registering sniffer
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/ControlEnvProxyServer.sock
netsimd I 01-02 11:42:07.397 rust_main.rs:94 - netsim artifacts path: /tmp/android-luyahan/netsimd
netsimd I 01-02 11:42:07.397 rust_main.rs:97 - NetsimdArgs {
    fd_startup_str: Some(
        "{\"devices\":[{\"name\":\"0.0.0.0:6520\",\"chips\":[{\"kind\":\"BLUETOOTH\",\"fdIn\":85,\"fdOut\":86}]}]}",
    ),
    no_cli_ui: false,
    no_web_ui: false,
    pcap: false,
    disable_address_reuse: false,
    hci_port: Some(
        7300,
    ),
    connector_instance: None,
    instance: Some(
        1,
    ),
    logtostderr: false,
    dev: false,
    vsock: None,
    config: None,
    test_beacons: false,
    no_test_beacons: false,
    no_shutdown: false,
    version: false,
}
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=30, out=29
KeyMint Rust TA running with infile=36, outfile=35, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
GUEST_UBOOT_VERSION: 2023.04-gc2f63e06296a-ab11193194 (Dec 05 2023 - 03:08:51 +0000)
U-Boot 20
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=30, out=29
KeyMint Rust TA running with infile=36, outfile=35, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
GUEST_KERNEL_VERSION: 6.6.0-mainline-gc387c4e84ce5-ab11047873 (build-user@build-host) (Android (10742551, +pgo, +bolt, +lto, +mlgo, based on r498229b) clang version 17.0.4 (https://android.googlesource.com/toolchain/llvm-project e34ed7d63863b45858e74126edaa738e75887800), LLD 17.0.4) #1 SMP PREEMPT Wed Nov  1 22:15:43 UTC 2023
] Linux version

Setting boot_info to BootInfo { verified_boot_key: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], device_boot_locked: false, verified_boot_state: Unverified, verified_boot_hash: [130, 179, 160, 211, 179, 248, 219, 233, 176, 11, 246, 206, 91, 46, 220, 209, 116, 142, 32, 250, 113, 159, 184, 230, 219, 152, 141, 179, 21, 100, 43, 199], boot_patchlevel: 20231205 }
Setting hal_info to HalInfo { os_version: 140000, os_patchlevel: 202312, vendor_patchlevel: 20231205 }
Set aidl_version to V3
Setting attestation IDs directly
Setting HMAC key from 1 shared secret parameters
transport message failed, response body: device '0.0.0.0:6520' not found
Fetching secret with slot ID: 1
transport message failed, response body: device '0.0.0.0:6520' not found
GUEST_BUILD_FINGERPRINT: generic/aosp_cf_riscv64_phone/vsoc_riscv64:VanillaIceCream/MAIN.82b3a0d3/11255113:userdebug/test-keys
Setting HMAC key from 1 shared secret parameters
transport message failed, response body: device '0.0.0.0:6520' not found
```

gongwuchun, 2024-01-04:
you could have a try this patch: https://android-review.googlesource.com/c/device/google/cuttlefish/+/2897661

gongwuchun, 2024-01-06:
Fixed by https://android-review.googlesource.com/c/device/google/cuttlefish/+/2899827 and it has been merged.

closed 2024-01-08 by enh-google