google/android-riscv64 #134: Can we have linux_musl-riscv64 support in prebuilts/build-tools?

gongwuchun, 2024-01-17:


enh-google, 2024-01-17:
what for?

gongwuchun, 2024-01-18:
> what for?
I just wonder if we can build riscv64 host package which rely on linux_musl-riscv64 build-tools then we are able to try riscv64 cuttlefish on risc-v hardware and no need to adapt for HALs, its performance may not good without KVM support(no H extension) but at least we can run riscv64 cuttlefish on risc-v hardware first.

btw, it there any guide how to enable linux_musl-riscv64 support in prebuilts/build-tools?

enh-google, 2024-01-18:
> at least we can run riscv64 cuttlefish on risc-v hardware first

but _why_? this isn't something we're likely to look at until there's hardware that makes that interesting from a performance perspective. and it would be a _lot_ of work, because you'll need everything --- soong, clang, jdk, ...

we've only just added this for arm64, a decade after devices shipped, so i have no reason to expect to see this before the mid-2030s!

closed 2024-01-18 by enh-google