google/android-riscv64 #137: Cuttlefish run failed with Enabling --gpu_mode=gfxstream

LeviMarvin, 2024-02-05:
Host details:
```
CPU: Intel Core i7 13700H
GPU: (prime-select on-demand) intel iGPU + NVIDIA RTX 4060 Laptop
```
Boot command: 
```
__NV_PRIME_RENDER_OFFLOAD=1 __VK_LAYER_NV_optimus=NVIDIA_only __GLX_VENDOR_LIBRARY_NAME=nvidia HOME=$CUTTLEFISH_HOME  launch_cvd --gpu_mode=auto --cpus 6 --memory_mb 8192
```
But If run the command without `--cpus 6 --memory_mb 8192`, then it can be booted with default cpus and gpus.

Boot log:
```text
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services. You can adjust the permission for sending
diagnostic information to Google, such as crash reports and usage
data from this Android Virtual Device, at any time by running
"launch_cvd -report_anonymous_usage_stats=n"
===================================================================

launch_cvd I 02-05 22:55:40 15351 15351 main.cc:324] Host changed from last run: 0
assemble_cvd D 02-05 22:55:40 15808 15808 fetcher_config.cpp:224] Could not find file ending in kernel
assemble_cvd D 02-05 22:55:40 15808 15808 fetcher_config.cpp:224] Could not find file ending in initramfs.img
assemble_cvd I 02-05 22:55:40 15808 15808 config_flag.cpp:153] Launching CVD using --config='phone'.
assemble_cvd D 02-05 22:55:40 15808 15808 subprocess.cpp:333] Started (pid: 15811): /home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 02-05 22:55:40 15808 15808 subprocess.cpp:335] /home/levi/workspace/projects/aosp/14/out/target/product/dotcat/boot.img
assemble_cvd D 02-05 22:55:41 15808 15808 subprocess.cpp:333] Started (pid: 15868): /home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/unpack_bootimg
assemble_cvd D 02-05 22:55:41 15808 15808 subprocess.cpp:335] --boot_img
assemble_cvd D 02-05 22:55:41 15808 15808 subprocess.cpp:335] /home/levi/workspace/projects/aosp/14/out/target/product/dotcat/boot.img
assemble_cvd D 02-05 22:55:41 15808 15808 subprocess.cpp:335] --out
assemble_cvd D 02-05 22:55:41 15808 15808 subprocess.cpp:335] /tmp/gMFR5F
NVVM compilation failed: 1
Failed to fully check if driver has issue when sampler2D has a 'lowp' precision qualifier
NVVM compilation failed: 1
Failed to fully check if driver has issue when sampler2D has a 'lowp' precision qualifier
GPU auto mode: detected prerequisites for accelerated rendering support.
Enabling --gpu_mode=gfxstream.
Path for instance UDS: /tmp/cf_avd_1000
The following files contain useful debugging information:
  Logcat output: /home/levi/workspace/cvd_home/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /home/levi/workspace/cvd_home/cuttlefish/instances/cvd-1/kernel.log
  Point your browser to https://localhost:8443 to interact with the device.
  Launcher log: /home/levi/workspace/cvd_home/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/levi/workspace/cvd_home/cuttlefish/instances/cvd-1/cuttlefish_config.json
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/adb_connector
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/socket_vsock_proxy
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/socket_vsock_proxy
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/tcp_connector
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/config_server
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/echo_server
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/gnss_grpc_proxy
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/logcat_receiver
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/kernel_log_monitor
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/metrics
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/log_tee
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/process_restarter
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/openwrt_control_server
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/log_tee
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/process_restarter
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/socket_vsock_proxy
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/socket_vsock_proxy
failed to read from socket, retry.
/home/levi/workspace/projects/aosp/14/out/host/linux-x86/bin/secure_env
Server listening on unix:/tmp/cf_avd_1000/cvd-1/grpc_socket/EchoServer.sock
[... 129 lines cut ...]
[2024/02/05 23:01:13:1333] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/02/05 23:01:13:1333] N: __lws_lc_tag:  ++ [vh|0|default|127.0.0.1|127.0.0.1|0] (1)
[2024/02/05 23:01:13:1339] N: [vh|0|default|127.0.0.1|127.0.0.1|0]: lws_socket_bind: source ads 127.0.0.1
[2024/02/05 23:01:13:1339] N: __lws_lc_tag:  ++ [wsi|1|listen|default|127.0.0.1|41547] (2)
(adm_helpers.cc:48): Unable to access speaker.
GUEST_UBOOT_VERSION: 2023.01-21885-gb3cc1cd24d (Apr 13 2023 - 16:27:56 +0000)
U-Boot 20
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=72, out=71
KeyMint Rust TA running with fd_in=78, fd_out=77, security_level=TrustedEnvironment
GUEST_KERNEL_VERSION: 6.1.23-android14-4-00257-g7e35917775b8-ab9964412 (build-user@build-host) (Android (9796371, based on r487747) clang version 17.0.0 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.0) #1 SMP PREEMPT Mon Apr 17 20:50:58 UTC 2023
] Linux version
Metrics posted to ClearCut
GUEST_BUILD_FINGERPRINT: Hidium/aosp_dotcat/dotcat:14/UQ1A.240105.004.A1.e1af3375/eng.levi.20240113.231311:userdebug/test-keys
init: starting service 'adbd'...
proxy_adb] Start event (5) received. Starting proxy
proxy_adb] From: fd: 46
proxy_adb] To: vsock: 3:5555
VIRTUAL_DEVICE_DISPLAY_POWER_MODE_CHANGED
VIRTUAL_DEVICE_DISPLAY_POWER_MODE_CHANGED
VIRTUAL_DEVICE_BOOT_STARTED
VIRTUAL_DEVICE_BOOT_COMPLETED
Virtual device booted successfully
VIRTUAL_DEVICE_NETWORK_MOBILE_CONNECTED
```

prashanthswami, 2024-02-15:
You may need to share your launcher.log for us to confirm (the location of that log is printed above in the output), but I suspect if your virtual device is running for a while and then crashing, is your system perhaps getting tight on memory?

The recommendations for the `cpus` flag and the `memory_mb` flag are not hard-and-fast rules, they are only some suggested values we've provided as we've found the defaults can be quite slow on the average desktop. You can scale them up and down depending on your host resources.

LeviMarvin, 2024-02-24:
@prashanthswami I'm sorry for I can't provide the log. I have turned to Windows 11 platform with WSL2. Is there a way to run Cuttlefish on WSL2? And the UI provided by WebRTC has some delay for interaction.