google/android-riscv64 #145: Build instructions for the prebuilt QEMU with rutabaga/gfxstream support ?

Ruinland, 2024-04-16:
The instructions on [crsovm doc](https://crosvm.dev/book/appendix/rutabaga_gfx.html) seems kinda outdated.

I am having linking issues during the final stage when building QEMU : 
```
/usr/bin/ld: /data_ssd_2t/ruinland_aosp/cuttlefish/qemu_rutabaga/build/deps/prefix/lib/librutabaga_gfx_ffi.so: undefined reference to `gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::tr
ansformToGuestMemoryRequirements(VkMemoryRequirements*) const'
/usr/bin/ld: /data_ssd_2t/ruinland_aosp/cuttlefish/qemu_rutabaga/build/deps/prefix/lib/librutabaga_gfx_ffi.so: undefined reference to `gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::Em
ulatedPhysicalDeviceMemoryProperties(VkPhysicalDeviceMemoryProperties const&, unsigned int, gfxstream::host::FeatureSet const&)'
/usr/bin/ld: /data_ssd_2t/ruinland_aosp/cuttlefish/qemu_rutabaga/build/deps/prefix/lib/librutabaga_gfx_ffi.so: undefined reference to `gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::ge
tHostMemoryInfoFromGuestMemoryTypeIndex(unsigned int) const'
```

I doubly checked with `libgfxstream_backend.a`, the symbols are indeed undefined : 
```
0000000000000000 W gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::getGuestMemoryProperties() const
                 U gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::transformToGuestMemoryRequirements(VkMemoryRequirements*) const
                 U gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::getHostMemoryInfoFromGuestMemoryTypeIndex(unsigned int) const
...
0000000000009f58 T gfxstream::vk::importExtMemoryHandleToVkColorBuffer(unsigned int, unsigned int, int)
                 U gfxstream::vk::EmulatedPhysicalDeviceMemoryProperties::EmulatedPhysicalDeviceMemoryProperties(VkPhysicalDeviceMemoryProperties const&, unsigned int, gfxstream::host::Feature
Set const&)
```

And these methods are supposedly to be part of gfxstream.

Is there a write up of details (including proper commit hashes) for building the QEMU with rubataga/gfxstream ?
I tried the ones from : 
https://linaro.atlassian.net/wiki/spaces/ORKO/pages/28985622530/Building+QEMU+with+virtio-gpu+and+rutabaga+gfx
Yet the commit hashes are quite old and fail to build.

Kamboh223, 2024-04-18:
[image]

enh-google, 2024-04-18:
(all we have atm is device/google/cuttlefish_vmm/qemu --- this has the README and build scripts, but there's no nice step by step, as the build process is finicky.)

Ruinland, 2024-05-03:
Thanks @enh-google , I managed to get the QEMU built with gfxstream by taking a look at the build script.

However, the problem I am encountering now is that - - the output of bootanimation is fine.
Even though the cvd-launcher constantly spills this error message : 
```
E4018 14:05:07.989282 454871 FrameBuffer.cpp:3303] Failed to find process owned resources for puid 13996581337902
```

And once the Android init brings up SystemUI, the cvd launcher floods with this error every single ms all over the place : 
```
qemu-system-riscv64 : CHECK failed in rubataga_cmd_resource_flush() ../qemu/hw/display/virtui-gpu-rubataga.c:234
```
in the meantime, the display is filed with noises :
[image]
 
====

Shall I rename this issue or close it and then open a new one ?

enh-google, 2024-05-04:
> Shall I rename this issue or close it and then open a new one ?

you might be better off on the mailing list, since i don't know how many people other than me and prashanthsw subscribe to all the issues, and we don't use this configuration... you're more likely to get help from another non-Googler trying to do the same thing than you are from us :-)

enh-google, 2024-05-04:
btw, did you see that we [well, "prashanthsw"] updated to qemu 9? i don't remember why you were trying to build your own qemu, but if you just wanted a newer one...

Ruinland, 2024-05-06:
@enh-google 
> we don't use this configuration
You mean gfxstream in general or QEMU 8.2 build w/ rubataga ?

Thanks a lot for the tips. I will write up a report and asked in the mailing list.
Yet I think I will keep this issue for non-RVIA members.

= = = =

>  i don't remember why you were trying to build your own qemu 

Well, vendor flavored QEMU is always a thing.
And I believe rubataga was once an offshoot, right 🤣