google/android-riscv64 #146: Run 'launch_cvd' can't start cuttlefish device

lpkm, 2024-07-17:
:lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

When run the commond 'launch_cvd'，The terminal outputs the following：
===================================================================
NOTICE:

We collect usage statistics in accordance with our
Content Licenses (https://source.android.com/setup/start/licenses),
Contributor License Agreement (https://cla.developers.google.com/),
Privacy Policy (https://policies.google.com/privacy) and
Terms of Service (https://policies.google.com/terms).
===================================================================

launch_cvd I 07-17 15:26:52  5671  5671 main.cc:183] Host changed from last run: 0
assemble_cvd D 07-17 15:26:52  6124  6124 fetcher_config.cpp:212] Could not find file ending in kernel
assemble_cvd D 07-17 15:26:52  6124  6124 fetcher_config.cpp:212] Could not find file ending in initramfs.img
assemble_cvd I 07-17 15:26:52  6124  6124 config_flag.cpp:148] Launching CVD using --config='auto'.
assemble_cvd D 07-17 15:26:52  6124  6124 subprocess.cpp:335] Started (pid: 6127): /home/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 07-17 15:26:52  6124  6124 subprocess.cpp:337] /home/out/target/product/vsoc_x86/boot.img
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

GPU auto mode: did not detect prerequisites for accelerated rendering support, enabling --gpu_mode=guest_swiftshader.
The following files contain useful debugging information:
  Serial console is disabled; use -console=true to enable it.
  Kernel log: /home/ball/cuttlefish/instances/cvd-1/kernel.log
  Logcat output: /home/ball/cuttlefish/instances/cvd-1/logs/logcat
  Launcher log: /home/ball/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/ball/cuttlefish/instances/cvd-1/cuttlefish_config.json
  Instance environment: /home/ball/.cuttlefish.sh
/home/out/host/linux-x86/bin/adb_connector
/home/out/host/linux-x86/bin/socket_vsock_proxy
/home/out/host/linux-x86/bin/bt_connector
/home/out/host/linux-x86/bin/config_server
/home/out/host/linux-x86/bin/kernel_log_monitor
/home/out/host/linux-x86/bin/logcat_receiver
/home/out/host/linux-x86/bin/metrics
/home/out/host/linux-x86/bin/log_tee
/home/out/host/linux-x86/bin/root-canal
/home/out/host/linux-x86/bin/secure_env
/home/out/host/linux-x86/bin/tombstone_receiver
/home/out/host/linux-x86/bin/crosvm
/home/out/host/linux-x86/bin/log_tee
/home/out/host/linux-x86/bin/log_tee
/home/out/host/linux-x86/bin/wmediumd
failed to read from socket, retry.
/home/out/host/linux-x86/bin/log_tee
/home/out/host/linux-x86/bin/crosvm
/home/out/host/linux-x86/bin/modem_simulator
/home/out/host/linux-x86/bin/operator_proxy
/home/out/host/linux-x86/bin/webRTC
Start modem simulator, server_fds: 76, Sim type: normal
[ERROR:external/crosvm/src/main.rs:2884] crosvm has exited with error: failed to lock disk image: Resource temporarily unavailable (os error 11)
Detected unexpected exit of monitored subprocess /home/out/host/linux-x86/bin/crosvm
Subprocess /home/out/host/linux-x86/bin/crosvm (6682) has exited with exit code 1
WaylandServer running on socket 18
[2024/07/17 15:26:55:8851] N: lws_create_context: LWS: 4.3.0, NET CLI SRV H1 H2 WS ConMon IPv6-absent
[2024/07/17 15:26:55:8852] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/07/17 15:26:55:8852] N: __lws_lc_tag:  ++ [vh|0|default|127.0.0.1|127.0.0.1|0] (1)
[2024/07/17 15:26:55:8872] N: [vh|0|default|127.0.0.1|127.0.0.1|0]: lws_socket_bind: source ads 127.0.0.1
[2024/07/17 15:26:55:8872] N: __lws_lc_tag:  ++ [wsi|1|listen|default|127.0.0.1|36753] (2)
(adm_helpers.cc:48): Unable to access speaker.
GUEST_UBOOT_VERSION: 2022.01-15068-g240b124907 (Apr 14 2022 - 14:14:27 +0000)
U-Boot 20
GUEST_KERNEL_VERSION: 5.15.41-android13-8-00029-g24d27dff64c4-ab9178016 (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Fri Oct 14 19:05:44 UTC 2022
] Linux version
transport message failed, response body:
transport message failed, response body:
transport message failed, response body:

and the adb devices is offline. Please help me resolve this problem, thanks.

enh-google, 2024-07-17:
this looks like the [first] error?

> [ERROR:external/crosvm/src/main.rs:2884] crosvm has exited with error:
failed to lock disk image: Resource temporarily unavailable (os error 11)

maybe you have a lock held from a previous failure? (which might mean that
this isn't even your real problem.)

or you just already have it running?

lpkm, 2024-07-18:
Thanks for the reply. 
Yes, you're right, I've noticed the problem and after running the stop_cvd instruction and then re-running the launch_cvd instruction the exception doesn't occur anymore.
My main problem is that the cuttlefish emulator, which is launched via the launch_cvd command, shows offline when viewing the device via adb devices, which has been bugging me for quite some time. The main reference to the Android website to install the cuttlefish environment, but as soon as you run cuttlefish can not start properly, adb devices show offline and print ‘the transport message failed, response body:’ 
Android official website: https://source.android.google.cn/docs/devices/cuttlefish/get-started?hl=nl&authuser=5

The terminal outputs the following：
===================================================================
NOTICE:

We collect usage statistics in accordance with our
Content Licenses (https://source.android.com/setup/start/licenses),
Contributor License Agreement (https://cla.developers.google.com/),
Privacy Policy (https://policies.google.com/privacy) and
Terms of Service (https://policies.google.com/terms).
===================================================================

launch_cvd I 07-18 13:05:24 1295133 1295133 main.cc:183] Host changed from last run: 0
assemble_cvd D 07-18 13:05:25 1295581 1295581 fetcher_config.cpp:212] Could not find file ending in kernel
assemble_cvd D 07-18 13:05:25 1295581 1295581 fetcher_config.cpp:212] Could not find file ending in initramfs.img
assemble_cvd I 07-18 13:05:25 1295581 1295581 config_flag.cpp:148] Launching CVD using --config='auto'.
assemble_cvd D 07-18 13:05:25 1295581 1295581 subprocess.cpp:335] Started (pid: 1295584): /home/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 07-18 13:05:25 1295581 1295581 subprocess.cpp:337] /home/out/target/product/vsoc_x86/boot.img
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

GPU auto mode: did not detect prerequisites for accelerated rendering support, enabling --gpu_mode=guest_swiftshader.
The following files contain useful debugging information:
  Serial console is disabled; use -console=true to enable it.
  Kernel log: /home/ball/cuttlefish/instances/cvd-1/kernel.log
  Logcat output: /home/ball/cuttlefish/instances/cvd-1/logs/logcat
  Launcher log: /home/ball/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/ball/cuttlefish/instances/cvd-1/cuttlefish_config.json
  Instance environment: /home/ball/.cuttlefish.sh
/home//out/host/linux-x86/bin/adb_connector
/home//out/host/linux-x86/bin/socket_vsock_proxy
/home//out/host/linux-x86/bin/bt_connector
/home//out/host/linux-x86/bin/config_server
/home//out/host/linux-x86/bin/kernel_log_monitor
/home//out/host/linux-x86/bin/logcat_receiver
/home//out/host/linux-x86/bin/metrics
/home//out/host/linux-x86/bin/log_tee
/home//out/host/linux-x86/bin/root-canal
/home//out/host/linux-x86/bin/secure_env
/home//out/host/linux-x86/bin/tombstone_receiver
/home//out/host/linux-x86/bin/crosvm
/home//out/host/linux-x86/bin/log_tee
/home//out/host/linux-x86/bin/log_tee
/home//out/host/linux-x86/bin/wmediumd
failed to read from socket, retry.
/home//out/host/linux-x86/bin/log_tee
/home//out/host/linux-x86/bin/crosvm
/home//out/host/linux-x86/bin/modem_simulator
/home//out/host/linux-x86/bin/operator_proxy
/home//out/host/linux-x86/bin/webRTC
Start modem simulator, server_fds: 76, Sim type: normal
WaylandServer running on socket 19
[2024/07/18 13:05:29:3158] N: lws_create_context: LWS: 4.3.0, NET CLI SRV H1 H2 WS ConMon IPv6-absent
[2024/07/18 13:05:29:3160] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2024/07/18 13:05:29:3160] N: __lws_lc_tag:  ++ [vh|0|default|127.0.0.1|127.0.0.1|0] (1)
[2024/07/18 13:05:29:3177] N: [vh|0|default|127.0.0.1|127.0.0.1|0]: lws_socket_bind: source ads 127.0.0.1
[2024/07/18 13:05:29:3178] N: __lws_lc_tag:  ++ [wsi|1|listen|default|127.0.0.1|38543] (2)
(adm_helpers.cc:48): Unable to access speaker.
GUEST_UBOOT_VERSION: 2022.01-15068-g240b124907 (Apr 14 2022 - 14:14:27 +0000)
U-Boot 20
GUEST_KERNEL_VERSION: 5.15.41-android13-8-00029-g24d27dff64c4-ab9178016 (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Fri Oct 14 19:05:44 UTC 2022
] Linux version
transport message failed, response body:
transport message failed, response body:
transport message failed, response body:
transport message failed, response body:

enh-google, 2024-07-18:
> <google/android-riscv64/issues/146/2235392666 ***@***.***

yeah, it's probably worth making sure you can run a regular x86-64
cuttlefish before trying riscv64 --- that way you can make sure you've done
all the cuttlefish setup correctly before worrying about riscv64-specific
problems.

lpkm, 2024-07-19:
When I run the run it shows adb devices as offline and after printing 'transport message failed, response body:' adb devices has no devices and after a few seconds it passes again. After a few seconds, adb devices can be viewed as offline again, and then the cycle repeats. I'm following the configurations from the official website exactly, what could be the cause of this problem?

prashanthswami, 2024-08-10:
When CVD is running, you'll see it output some files it's writing to - share your kernel and logcat from:

```
Kernel log: /home/ball/cuttlefish/instances/cvd-1/kernel.log
Logcat output: /home/ball/cuttlefish/instances/cvd-1/logs/logcat
``` 

If the device is booting but ADB isn't connecting, that could be a number of issues:

1. You might be trying to connect a bit too quickly - remember that RISC-V emulation on your X86 machine is going to be *very* slow, you may need to wait a few minutes before ADB is available.
2. Something might be blocking boot - this is unlikely because there are continuous boot tests run at ToT, but there's a possibility you might be at some bad CL as your sync point.

We should also note that if you want to do an equivalent test on the x86_64 image, you'll need to pass the additional flag `--vm_manager=qemu_cli`. RISC-V Cuttlefish uses QEMU by default, but X86_64 Cuttlefish uses CrosVM by default, so without this flag, you're not testing similar launch flows.

lpkm, 2024-08-14:
Thank you for your answer!

When I use the -- vm_manager=qemu_cli property to start cuttlefish, I encounter a new issue where it prompts me that port 7100 is occupied. But when I used the netstat - tulpn | grep 7100 command for filtering, I did not find any processes using this port. 
The startup command I used is: 
`launch_cvd` -- startw_ebrtc=true -- vm_manager=qemu_cli console=true.`

The following is the abnormal information printed:

launch_cvd I 08-14 10:04:18 2571579 2571579 main.cc:183] Host changed from last run: 0
assemble_cvd D 08-14 10:04:18 2572027 2572027 fetcher_config.cpp:212] Could not find file ending in kernel
assemble_cvd D 08-14 10:04:18 2572027 2572027 fetcher_config.cpp:212] Could not find file ending in initramfs.img
assemble_cvd I 08-14 10:04:18 2572027 2572027 config_flag.cpp:148] Launching CVD using --config='auto'.
assemble_cvd D 08-14 10:04:18 2572027 2572027 subprocess.cpp:335] Started (pid: 2572030): /home/ball/workspace/A13_r020_lxc/QSSI.13/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 08-14 10:04:18 2572027 2572027 subprocess.cpp:337] /home/ball/workspace/A13_r020_lxc/QSSI.13/out/target/product/vsoc_x86/boot.img
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

GPU auto mode: did not detect prerequisites for accelerated rendering support, enabling --gpu_mode=guest_swiftshader.
The following files contain useful debugging information:
  To access the console run: screen /home/ball/cuttlefish/instances/cvd-1/console
  Kernel log: /home/ball/cuttlefish/instances/cvd-1/kernel.log
  Logcat output: /home/ball/cuttlefish/instances/cvd-1/logs/logcat
  Launcher log: /home/ball/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/ball/cuttlefish/instances/cvd-1/cuttlefish_config.json
  Instance environment: /home/ball/.cuttlefish.sh
Port 7100 Bind failed (Address already in use)
Check failed: result.ok() Received `false`
Address already in use
  at device/google/cuttlefish/host/commands/run_cvd/launch_streamer.cpp:128
  in virtual Result<void> cuttlefish::(anonymous namespace)::StreamerSockets::ResultSetup()
  for CF_EXPECT(touch_socket->IsOpen())
Setup failed for StreamerSockets
  at device/google/cuttlefish/host/libs/config/feature.cpp:57
  in static Result<void> cuttlefish::SetupFeature::RunSetup(const std::vector<SetupFeature *> &)
  for CF_EXPECT(feature->ResultSetup())
Received error
  at device/google/cuttlefish/host/commands/run_cvd/main.cc:208
  in Result<void> cuttlefish::RunCvdMain(int, char **)
  for CF_EXPECT(SetupFeature::RunSetup(features))
launch_cvd E 08-14 10:04:20 2571579 2571579 subprocess.cpp:160] Subprocess 2572567 was interrupted by a signal: 6
launch_cvd E 08-14 10:04:20 2571579 2571579 main.cc:252] run_cvd returned -1

prashanthswami, 2024-08-14:
I don't believe you can use `--start_webrtc` and `--vm_manager=qemu_cli` together, the latter doesn't have WebRTC support. When using QEMU, you'll have to use VNC instead. You'll need to install any VNC viewer and use it to open the address `localhost:6444`.

In regards to you not seeing the port being used, I'd double check if you're `sudo`-ing that command - processes are generally not wrong about a port failing to open and *something* is definitely holding it open.

closed 2025-01-03 by lpkm