google/android-riscv64 #152: cmake: add riscv64 android to upstream?

truboxl, 2024-11-20:
Is it time to add riscv64-linux-android to cmake? I try building projects with NDK r27c but cmake seems to fall short of support.
I patch the cmake files myself to add the missing support.
```diff
--- a/Modules/Platform/Android/abi-riscv64-Clang.cmake
+++ b/Modules/Platform/Android/abi-riscv64-Clang.cmake
@@ -0,0 +1 @@
+include(Platform/Android/abi-common-Clang)
--- a/Modules/Platform/Android-Determine.cmake
+++ b/Modules/Platform/Android-Determine.cmake
@@ -353,6 +353,10 @@
   set(NDK_ABI_x86_64_ARCH              "x86_64")
   set(NDK_ABI_x86_64_TRIPLE            "x86_64-linux-android")
   set(NDK_ABI_x86_64_LLVM_TRIPLE       "x86_64-none-linux-android")
+  set(NDK_ABI_riscv64_PROC             "riscv64")
+  set(NDK_ABI_riscv64_ARCH             "riscv64")
+  set(NDK_ABI_riscv64_TRIPLE           "riscv64-linux-android")
+  set(NDK_ABI_riscv64_LLVM_TRIPLE      "riscv64-none-linux-android")
 
   set(NDK_PROC_aarch64_ABI "arm64-v8a")
   set(NDK_PROC_armv7-a_ABI "armeabi-v7a")
@@ -362,13 +366,15 @@
   set(NDK_PROC_mips_ABI    "mips")
   set(NDK_PROC_mips64_ABI  "mips64")
   set(NDK_PROC_x86_64_ABI  "x86_64")
+  set(NDK_PROC_riscv64_ABI "riscv64")
 
-  set(NDK_ARCH_arm64_ABI  "arm64-v8a")
-  set(NDK_ARCH_arm_ABI    "armeabi")
-  set(NDK_ARCH_mips_ABI   "mips")
-  set(NDK_ARCH_mips64_ABI "mips64")
-  set(NDK_ARCH_x86_ABI    "x86")
-  set(NDK_ARCH_x86_64_ABI "x86_64")
+  set(NDK_ARCH_arm64_ABI   "arm64-v8a")
+  set(NDK_ARCH_arm_ABI     "armeabi")
+  set(NDK_ARCH_mips_ABI    "mips")
+  set(NDK_ARCH_mips64_ABI  "mips64")
+  set(NDK_ARCH_x86_ABI     "x86")
+  set(NDK_ARCH_x86_64_ABI  "x86_64")
+  set(NDK_ARCH_riscv64_ABI "riscv64")
 endif()
 
 # Validate inputs.
```

enh-google, 2024-11-20:
lol that mips/mips64 are in that list...

enh-google, 2025-06-23:
@DanAlbert but this doesn't seem unreasonable, given that ndk-build already has support?

enh-google, 2025-06-23:
(and please feel free to remove mips/mips64 while you're there!)