google/android-riscv64 #154: cuttlefish安装失败

HaPo-yishui, 2024-12-10:
git clone https://github.com/google/android-cuttlefish
cd android-cuttlefish
tools/buildutils/build_packages.sh

.....
debian/rules build
dh build --with=config-package
   dh_update_autotools_config
   dh_autoreconf
   debian/rules override_dh_auto_build
make[1]: Entering directory '/home/ubuntu/riscv/Cuttlefish/android-cuttlefish/base'
cd cvd && bazel build --linkopt="-Wl,--build-id=sha1" cuttlefish/host/commands/cvd:cvd --spawn_strategy=local
Starting local Bazel server and connecting to it...
WARNING: For repository 'boringssl', the root module requires module version boringssl@0.0.0-20240126-22d349c, but got boringssl@0.20241024.0 in the resolved dependency graph. Please update the version in your MODULE.bazel or set --check_direct_dependencies=off
WARNING: For repository 'googletest', the root module requires module version googletest@1.14.0, but got googletest@1.15.2 in the resolved dependency graph. Please update the version in your MODULE.bazel or set --check_direct_dependencies=off
WARNING: For repository 'protobuf', the root module requires module version protobuf@21.7, but got protobuf@29.0 in the resolved dependency graph. Please update the version in your MODULE.bazel or set --check_direct_dependencies=off
WARNING: For repository 'zlib', the root module requires module version zlib@1.3.1, but got zlib@1.3.1.bcr.3 in the resolved dependency graph. Please update the version in your MODULE.bazel or set --check_direct_dependencies=off
WARNING: Couldn't auto load rules or symbols, because no dependency on module/repository 'rules_android' found. This will result in a failure if there's a reference to those rules or symbols.
INFO: Analyzed target //cuttlefish/host/commands/cvd:cvd (139 packages loaded, 3682 targets configured).
ERROR: /home/ubuntu/riscv/Cuttlefish/android-cuttlefish/base/cvd/cuttlefish/host/commands/cvd/BUILD.bazel:51:11: Compiling cuttlefish/host/commands/cvd/cli/parser/cf_flags_validator.cpp failed: (Exit 1): clang failed: error executing CppCompile command (from target //cuttlefish/host/commands/cvd:libcvd) /usr/lib/llvm-14/bin/clang -U_FORTIFY_SOURCE -fstack-protector -Wall -Wthread-safety -Wself-assign -Wunused-but-set-parameter -Wno-free-nonheap-object -fcolor-diagnostics -fno-omit-frame-pointer ... (remaining 192 arguments skipped)
cuttlefish/host/commands/cvd/cli/parser/cf_flags_validator.cpp:19:10: error: module //cuttlefish/host/commands/cvd:libcvd does not depend on a module exporting 'google/protobuf/util/json_util.h'
#include <google/protobuf/util/json_util.h>
         ^
1 error generated.

Please help answer, thanks

Databean, 2024-12-10:
This is an issue with upstream github.com/google/android-cuttlefish. I believe it is introduced by https://github.com/google/android-cuttlefish/pull/815 bringing in a new version of `libcurl` which implicitly pulls in a new version of `protobuf`. It is fixed by https://github.com/google/android-cuttlefish/pull/818, but that PR is blocked on submit by our presubmit tests, which are broken by `bazel`'s migration from using `WORKSPACE` by default to `bzlmod` by default. The WIP PR https://github.com/google/android-cuttlefish/pull/820 attempts to fix our e2e tests, but I am still working on it.

Databean, 2024-12-11:
FYI, https://github.com/google/android-cuttlefish/pull/820 is merged, fixing this issue.

HaPo-yishui, 2024-12-12:
> FYI, [google/android-cuttlefish#820](https://github.com/google/android-cuttlefish/pull/820) is merged, fixing this issue.

Hi, Thank you very much for your reply.

closed 2024-12-12 by enh-google