google/android-riscv64 #155: What is the level of ART support in AOSP RISCV, Please help

HaPo-yishui, 2024-12-19:
What is the current state of support for ART in AOSP Riscv? Only JIT is supported, or AOT is also supported, and what is the degree of support for both, Google please help to reply.
Thanks

enh-google, 2024-12-20:
yes, both should work and pass all tests.

closed 2024-12-20 by enh-google

HaPo-yishui, 2024-12-21:
Hi，enh-google：

I'm sorry, maybe I didn't understand you.
If ART supports AOT compilation, with WITH_DEXPREOPT set to true, then the corresponding .odex file should be generated in the CPU architecture directory ( such as */oat/riscv64/*.odex), but it is not generated. 
Therefore, I cannot make a normal judgment on the support of AOT compilation for aosp based on riscv architecture. Looking forward to your reply.
Thanks a lot

HaPo-yishui, 2024-12-21:
Hi，enh-google：

I'm sorry, maybe I didn't understand you.
If ART supports AOT compilation, with WITH_DEXPREOPT set to true, then the corresponding .odex file should be generated in the CPU architecture directory ( such as */oat/riscv64/*.odex), but I find it is not generated. 
Therefore, I cannot make a normal judgment on the support of AOT compilation for aosp based on riscv architecture. Looking forward to your reply.
Thanks a lot

enh-google, 2025-01-08:
building aosp main with
```
    . build/envsetup.sh
    lunch aosp_cf_riscv64_phone-trunk_staging-userdebug
    m
```
leaves a lot of compiled oat files in a variety of directories:
```
    out/target/product/vsoc_riscv64/system/framework/riscv64/ (boot image)
    out/target/product/vsoc_riscv64/system/framework/oat/riscv64/ (system server, Mainline modules)
    out/target/product/vsoc_riscv64/system/{app,priv-app}/<app-name>/oat/riscv64/
    out/target/product/vsoc_riscv64/system_other/{product,system,system_ext}/{app,priv-app}/<app-name>/oat/riscv64/
    out/target/product/vsoc_riscv64/vendor/priv-app/<app-name>/oat/riscv64/
```
are you _sure_ you have `WITH_DEXPREOPT` set (and have done a full build with that)?

HaPo-yishui, 2025-01-13:
> building aosp main with
>
> ```
>     . build/envsetup.sh
>     lunch aosp_cf_riscv64_phone-trunk_staging-userdebug
>     m
> ```
>
> leaves a lot of compiled oat files in a variety of directories:
>
> ``` [...]

Tks, WITH_DEXPREOPT has been set to true:
[image]

Then a full build is done, a lot of compiled oat files in a variety of directories are generated:
[image]

Thank you for your assistance

enh-google, 2025-01-14:
> If ART supports AOT compilation, with WITH_DEXPREOPT set to true, then the corresponding .odex file should be generated in the CPU architecture directory ( such as /oat/riscv64/.odex), but I find it is not generated.

i think you're looking for the wrong files ... you want `.oat` files, not `.odex`.

HaPo-yishui, 2025-01-15:
> i think you're looking for the wrong files ... you want `.oat` files, not `.odex`.

 ===> 
I only see that the corresponding .oat files is generated in the framework directory, but not in other directories:

[image]

Can you give me the reason why the .oat files are not generated? TKS

enh-google, 2025-01-15:
the ART folks say:

Our oat files are both .oat (for boot images) and .odex (for everything else - for compatibility reasons). So, the screenshot with red arrows pointing to .odex files is correct. Those are indeed oat files.

Apps seem to be compiled without app images, therefore the search for "*.art" does not find compiled artifacts for apps.

Searching for "*.vdex" is an excellent way to find all `dex2oat` output locations. Each .oat and .odex file should have a corresponding .vdex file. However, apps compiled with `--compiler-filter=verify` shall have a .vdex file without .odex.

HaPo-yishui, 2025-01-17:
Thank you for your reply. There is no doubt about it. TKS