google/android-riscv64 #156: ART AOT Compilation support in AOSP RISCV, Please help

HaPo-yishui, 2024-12-23:
[image]

Hi，enh-google：

I'm sorry, maybe I didn't understand you.
If ART supports AOT compilation, with WITH_DEXPREOPT set to true, then the corresponding .odex file should be generated in the CPU architecture directory ( such as /oat/riscv64/.odex), but I find it is not generated.
Therefore, I cannot make a normal judgment on the support of AOT compilation for aosp based on riscv architecture. Looking forward to your reply.
Thanks a lot

HaPo-yishui, 2024-12-31:
Can anyone answer my question? Thank you very much. 
Looking forward to google's reply

enh-google, 2025-01-09:
duplicate of #155

closed 2025-01-09 by enh-google