google/android-riscv64 #157: android15-qpr1-release run cuttlefish encounter panic issue

doublezhan, 2025-01-10:
When use android15-qpr1-release branch aosp_cf_riscv64_phone-trunk_staging-userdebug product with build kernel image from common-android-mainline-riscv64 branch, then run on cuttlefish, we encounter below issue, how can i solved this issue?

[ 120.665948][ T1] init: starting service 'apexd-snapshotde'...
[ 120.774242][ T1] init: ... started service 'apexd-snapshotde' has pid 496
[ 120.775792][ T1] init: SVC_EXEC service 'apexd-snapshotde' pid 496 (uid 0 gid 1000+0 context default) started; waiting...
[ 120.776522][ T1] init: Command 'exec_start apexd-snapshotde' action=post-fs-data (/system/etc/init/hw/init.rc:1093) took 114ms and succeeded
[ 120.782674][ T1] init: Sending signal 9 to service 'odsign' (pid 493) process group...
[ 120.804462][ T1] libprocessgroup: Removed cgroup /sys/fs/cgroup/uid_0/pid_493
[ 120.809709][ T1] init: Control message: Processed ctl.stop for 'odsign' from pid: 493 (/system/bin/odsign)
[ 120.814551][ C0] Oops - illegal instruction [https://github.com//issues/1]
[ 120.814641][ C0] Modules linked in: virtio_snd(E) virtio_pmem(E) virtio_balloon(E) virt_wifi(E) vhci_hcd(E) usbip_core(E) system_heap(E) snd_aloop(E) rtc_test(E) pulse8_cec(E) mt76x2u(E) mt76x2_common(E) mt76x0u(E) mt76x02_usb(E) mt76x0_common(E) mt76x02_lib(E) mt76_usb(E) mt76(E) hci_vhci(E) gs_usb(E) can_dev goldfish_sync(OE) goldfish_pipe(OE) goldfish_battery(E) goldfish_address_space(OE) dummy_hcd(E) dummy_cpufreq(E) btusb(E) btbcm btrtl(E) btintel(E) zram bluetooth zsmalloc mac80211_hwsim(E) mac80211(E) cfg80211(E) rfkill libarc4 vmw_vsock_virtio_transport virtio_rng(E) virtio_pci virtio_pci_legacy_dev virtio_pci_modern_dev virtio_net(E) virtio_input(E) virtio_gpu(E) virtio_dma_buf(E) virtio_console virtio_blk net_failover(E) nd_virtio(E) failover(E)
[ 120.815787][ C0] CPU: 0 UID: 0 PID: 1 Comm: init Tainted: G OE 6.12.0-mainline-maybe-dirty #1
[ 120.815938][ C0] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[ 120.815964][ C0] Hardware name: riscv-virtio,qemu (DT)
[ 120.816077][ C0] epc : percpu_ref_put+0x118/0x11a
[ 120.816555][ C0] ra : percpu_ref_put+0x28/0x11a
[ 120.816579][ C0] epc : ffffffff80158fa2 ra : ffffffff80158eb2 sp : ffff8f800002bc70
[ 120.816601][ C0] gp : ffff8f800000d068 tp : ffffaf8080238000 t0 : ffffaf80802387ff
[ 120.816620][ C0] t1 : 0000000072ab2a2a t2 : 0000000000000000 s0 : ffff8f800002bc90
[ 120.816639][ C0] s1 : ffffaf808b344010 a0 : 7fffffffffffffff a1 : 8000000000000000
[ 120.816658][ C0] a2 : 2de9030836a50700 a3 : 000000000000000f a4 : 0000000000040000
[ 120.816678][ C0] a5 : 0000000000000008 a6 : 0000000000000001 a7 : 0000000000000002
[ 120.816696][ C0] s2 : ffffaf80899fb640 s3 : 0000000000000000 s4 : ffffaf8080b5dc00
[ 120.816714][ C0] s5 : ffffaf80899fb640 s6 : ffffaf80899fdde8 s7 : ffffffff823160e0
[ 120.816734][ C0] s8 : 0000000000000001 s9 : 0000000000000004 s10: 0000555573829a18
[ 120.816752][ C0] s11: 0000555573824ac0 t3 : 0000000000000000 t4 : 0000000000000000
[ 120.816770][ C0] t5 : 0000000000000040 t6 : ffffffffec35d120
[ 120.816788][ C0] status: 0000000200000120 badaddr: 0000000000000000 cause: 0000000000000002
[ 120.816868][ C0] [] percpu_ref_put+0x118/0x11a
[ 120.816952][ C0] [] cgroup_sk_free+0x2e/0x44
[ 120.816973][ C0] [] __sk_destruct+0x14a/0x1fc
[ 120.816990][ C0] [] __sk_free+0xee/0x124
[ 120.817004][ C0] [] sk_free+0x30/0x50
[ 120.817017][ C0] [] unix_release_sock+0x254/0x348
[ 120.817032][ C0] [] unix_release+0x3c/0x58
[ 120.817046][ C0] [] sock_close+0x54/0xe8
[ 120.817058][ C0] [] __fput+0x8a/0x256
[ 120.817072][ C0] [] __fput_sync+0x62/0x86
[ 120.817084][ C0] [] __riscv_sys_close+0x72/0xb4
[ 120.817095][ C0] [] syscall_handler+0x52/0x64
[ 120.817108][ C0] [] do_trap_ecall_u+0x98/0x184
[ 120.817121][ C0] [] _skip_scs+0x38/0x46
[ 120.817257][ C0] Code: ffc5 23b7 1731 839b 8d33 0363 0073 9002 9582 bfad (0000) 01a1
[ 120.817621][ C0] ---[ end trace 0000000000000000 ]---
[ 120.817920][ C0] Kernel panic - not syncing: Fatal exception in interrupt[ 120.818486][ C0] Kernel Offset: 0x0 from

enh-google, 2025-01-10:
shouldn't you also have a "unhandled signal" line here that shows what the illegal instruction was?
```
[ 120.814551][ C0] Oops - illegal instruction [https://github.com/https://github.com/google/android-riscv64/issues/1]
```
(you should probably have used ``` to quote all this to ensure you don't get weirdness like that fake github link too!)

doublezhan, 2025-01-11:
> shouldn't you also have a "unhandled signal" line here that shows what the illegal instruction was?
>
> ```
> [ 120.814551][ C0] Oops - illegal instruction [...]

[image]

enh-google, 2025-01-14:
fwiw, AOSP main built and booted fine for me today. you'll probably have to debug this locally to work out what the invalid instruction actually is.

doublezhan, 2025-02-21:
> fwiw, AOSP main built and booted fine for me today. you'll probably have to debug this locally to work out what the invalid instruction actually is.

When i switch clang toolchain to clang-r510928(build by common-android15-6.6 branch) , build common-android-mainline-riscv64 branch, then run android15-qpr1-release aosp_cf_riscv64_phone-trunk_staging-userdebug product on cuttlefish, boot is fine and not encounter illegal instruction issue.
which difference between clang-r510928(build by common-android15-6.6 branch) and clang-r536225(build by common-android-mainline-riscv64 branch)?
Why clang-r510928(build by common-android15-6.6 branch) is ok?

closed 2025-02-24 by enh-google