google/android-riscv64 #162: Structure accesses compiled with NDK r27 for RISC-V produce more instructions than expected

MaoHan001, 2025-04-03:
For example, the following code

typedef long unsigned int size_t;                                                                                                                                                                                  
                                                                                                                                                                                                                   
pixel_bgr copy_bgr(unsigned char *bitmap, int row, int col){                                                                                                                                                       
    size_t elem_size = 3;                                                                                                                                                                                          
    size_t index = 0;                                                                                                                                                                                              
    pixel_bgr bgr = { 0 };                                                                                                                                                                                         
                                                                                                                                                                                                                   
                                                                                                                                                                                                                   
    index = (row * 40 + col) * 3;                                                                                                                                                                                  
    bgr = *((pixel_bgr*)(bitmap + index));                                                                                                                                                                         
                                                                                                                                                                                                                   
    return bgr;                                                                                                                                                                                                    
}  

The generated code is as follows:
        .text                                                                                                                                                                                                      
        .attribute      4, 16                                                                                                                                                                                      
        .attribute      5, "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_v1p0_zicsr2p0_zba1p0_zbb1p0_zbs1p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvl128b1p0_zvl32b1p0_zvl64b1p0"                                 
        .file   "2.ii"                                                                                                                                                                                             
        .globl  _Z8copy_bgrPhii                 # -- Begin function _Z8copy_bgrPhii                                                                                                                                
        .p2align        1                                                                                                                                                                                          
        .type   _Z8copy_bgrPhii,@function                                                                                                                                                                          
_Z8copy_bgrPhii:                        # @_Z8copy_bgrPhii                                                                                                                                                         
        .cfi_startproc                                                                                                                                                                                             
# %bb.0:                                                                                                                                                                                                           
        sh2add  a1, a1, a1                                                                                                                                                                                         
        sh3add  a1, a1, a2                                                                                                                                                                                         
        sh1add  a1, a1, a1                                                                                                                                                                                         
        sext.w  a1, a1                                                                                                                                                                                             
        add     a0, a0, a1                                                                                                                                                                                         
        lbu     a1, 1(a0)                                                                                                                                                                                          
        lbu     a2, 0(a0)                                                                                                                                                                                          
        lbu     a0, 2(a0)                                                                                                                                                                                          
        slli    a1, a1, 8                                                                                                                                                                                          
        or      a1, a1, a2                                                                                                                                                                                         
        slli    a0, a0, 16                                                                                                                                                                                         
        or      a0, a0, a1                                                                                                                                                                                         
        ret                                                                                                                                                                                                        
.Lfunc_end0:                                                                                                                                                                                                       
        .size   _Z8copy_bgrPhii, .Lfunc_end0-_Z8copy_bgrPhii                                                                                                                                                       
        .cfi_endproc                                                                                                                                                                                               
                                        # -- End function                                                                                                                                                          
        .ident  "Android (dev, -pgo, -bolt, -lto, -mlgo, based on r522817) clang version 18.0.1 (https://android.googlesource.com/toolchain/llvm-project d8003a456d14a3deb8054cdaa529ffbf02d9b262)"                
        .section        ".note.GNU-stack","",@progbits   

Two lbu can be replaced by one lhu，like：
39400949      ldrb    w9, [x10, #0x2]
7940014a      ldrh    w10, [x10]
2a094140      orr     w0, w10, w9, lsl #16

enh-google, 2025-04-03:
yeah, "move stuff from build/soong into the clang driver itself" is on the to-do list, but no-one's working on it atm. (in particular because obviously there's still more stuff that needs to go into build/soong too.) but if alibaba or anyone else wants to upload llvm patches, i can find someone from the Android llvm team to approve them. @pirama-arumuga-nainar fyi.

(in the meantime, i'll add the CTS tests for the RISCV_HWPROBE_KEY_MISALIGNED_SCALAR_PERF/RISCV_HWPROBE_KEY_MISALIGNED_VECTOR_PERF distinction.)

enh-google, 2025-04-14:
actually, pirama points out that this has already been upstreamed, and removed from the build system in AOSP almost a year ago: https://android-review.googlesource.com/c/platform/build/soong/+/3149038

so just updating to a newer clang should fix this.

(i'll let him comment on whether r28 is actually new enough.)

pirama-arumuga-nainar, 2025-04-15:
https://github.com/llvm/llvm-project/pull/85704 added this in clang and is included in r28: https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/refs/heads/ndk-r28-release/clang-r530567c/clang_source_info.md.

closed 2025-04-15 by pirama-arumuga-nainar