google/android-riscv64 #164: Question about malloc performace

MaoHan001, 2025-04-08:
During our recent efforts to optimize system performance, we identified several memory allocation-related overheads. We observed varying performance advantages between jemalloc and Scudo across different scenarios:
1. Allocation of a small number of large memory chunks:
	◦	Scudo introduces additional overhead by invoking memset to initialize memory.
2. Loop-driven access to multiple fixed-size memory blocks allocated contiguously:
	◦	jemalloc allocates memory linearly in fixed sizes, allowing the processor to prefetch addresses efficiently (reducing cache misses), whereas Scudo’s address randomization degrades performance in this case.
3. Allocation of a massive number of small memory chunks:
	◦	jemalloc frequently uses mmap, leading to a number of pagefaults, while Scudo performs better here.

Ignoring security considerations, is there a way to configure Scudo to achieve optimal performance? (e.g. scudo includes a zero_contents configuration. )

enh-google, 2025-04-08:
@cferris1000 for alternative scudo configurations...

(there's some very high-level documentation at https://source.android.com/docs/security/test/scudo that we should probably update, but i think cferris has documentation for an easy-to-use prepackaged alternative configuration somewhere.)

cferris1000, 2025-04-08:
The best documentation for the config parameters of Scudo is:

[https://cs.android.com/android/platform/superproject/main/+/main:external/scudo/standalone/allocator_config.def](https://cs.android.com/android/platform/superproject/main/+/main:external/scudo/standalone/allocator_config.def)

In the case of turning off the zeroing, that is not a config option, but instead a flag that defaults to off. It is enabled by Android itself, and can be turned off if you change the bionic/libc/Android.bp and comment out the -DSCUDO_ZERO_CONTENTS line.

enh-google, 2025-04-08:
is there somewhere that talks about the low-end configuration? i don't think alibaba wants to manually tune things --- they just want the variant for low-end devices (like all existing risc-v hardware)...

cferris1000, 2025-04-08:
Not yet. The current low-end config in the Android tree is mostly experimental and I'm adding configurable parts to make it easier to use.

cferris1000, 2025-04-17:
If you do want to look at the current low memory config for scudo, it is here:

https://android.googlesource.com/platform/external/scudo/+/main/config/custom_scudo_config.h

The low memory config is named AndroidLowMemoryConfig. You can also enable this by default by modifying the blueprint file:

https://android.googlesource.com/platform/external/scudo/+/main/Android.bp#82

Add -DSCUDO_LOW_MEMORY to the cflags above that in the same cc_defaults definition. That will enable using that low memory config.

MaoHan001, 2025-04-18:
Thank you for sharing the information. Both links are very useful to us.

closed 2025-04-18 by MaoHan001